sheeriot/surveyor #16: Attribute Error on PackTrack and Server Error (500)

mkhatun1, 2024-04-02:
Hi, 
We were experiencing the Attribute Error, when trying to check this GlobalSat501 packet on Surveyor: 000db537076a397e this morning. Now we are getting Server Error (500). Please check the issue here. 

Thanks!

krisdthompson, 2024-04-02:
fixed an error on creating an empty dataframe (should be pd.Dataframe() )
Now gettting an error trying to normalize the results set
https://surveyor1.kttex.com/packTrack/packGraph/?endnode=54&start=2024-04-01T00%3A00&end=2024-04-02T16%3A53&submit=Submit
`Cannot convert non-finite values (NA or inf) to integer: Error while type casting for column 'frame_size'"]`

krisdthompson, 2024-04-03:
fixed more errors on geoview

krisdthompson, 2024-04-04:
resolved and deployed

closed 2024-04-04 by krisdthompson