sheeriot/surveyor #31: Read Timed Out Error in Fetching A month of Data

mkhatun1, 2024-04-15:
Getting a "Read Timed Out" Error while trying to pull a month of Data for Mueller S Daytona, FL. Here is the link: https://surveyor.everynet.com/performer/bucketDevicesReport/14/mdayt1/20240315T0500Z/20240415T0500Z/

krisdthompson, 2024-04-16:
This is the failed task (timeout).
https://surveyor.everynet.com/admin/django_celery_results/taskresult/280

krisdthompson, 2024-04-16:
Series of 20 second timeouts shown below:

Why timeout after 20 seconds?
```
surveyor_worker-1  | [2024-04-15 20:46:55,052: INFO/MainProcess] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[51147c09-843e-4bba-94cf-d35ef5969ee6] received
surveyor_worker-1  | [2024-04-15 20:47:15,238: INFO/ForkPoolWorker-2] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[51147c09-843e-4bba-94cf-d35ef5969ee6] succeeded in 20.185087068995927s: ('Failed: HTTPSConnectionPool(host=\'eastus-1.azure.cloud2.influxdata.com\', port=443): Read timed out. (read timeout=20.0)', {})
surveyor_worker-1  | [2024-04-15 21:19:12,561: INFO/MainProcess] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[99553447-f704-4888-a611-177721b22084] received
surveyor_worker-1  | [2024-04-15 21:19:32,780: INFO/ForkPoolWorker-2] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[99553447-f704-4888-a611-177721b22084] succeeded in 20.217246312997304s: ('Failed: HTTPSConnectionPool(host=\'eastus-1.azure.cloud2.influxdata.com\', port=443): Read timed out. (read timeout=20.0)', {})
surveyor_worker-1  | [2024-04-15 21:19:42,385: INFO/MainProcess] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[dda2cc91-39b4-43e9-abc2-179b20653ae8] received
surveyor_worker-1  | [2024-04-15 21:20:02,646: INFO/ForkPoolWorker-2] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[dda2cc91-39b4-43e9-abc2-179b20653ae8] succeeded in 20.260037083004136s: ('Failed: HTTPSConnectionPool(host=\'eastus-1.azure.cloud2.influxdata.com\', port=443): Read timed out. (read timeout=20.0)', {})
surveyor_worker-1  | [2024-04-15 21:20:17,628: INFO/MainProcess] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[4fd5dbd0-3e15-4a15-962b-e91ac114ef73] received
surveyor_worker-1  | [2024-04-15 21:20:37,791: INFO/ForkPoolWorker-2] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[4fd5dbd0-3e15-4a15-962b-e91ac114ef73] succeeded in 20.162173650984187s: ('Failed: HTTPSConnectionPool(host=\'eastus-1.azure.cloud2.influxdata.com\', port=443): Read timed out. (read timeout=20.0)', {})
surveyor_worker-1  | [2024-04-15 21:22:01,747: INFO/MainProcess] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[7fc448fe-d476-42c5-aa95-ff663bf9a84b] received
surveyor_worker-1  | [2024-04-15 21:22:21,936: INFO/ForkPoolWorker-2] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[7fc448fe-d476-42c5-aa95-ff663bf9a84b] succeeded in 20.187965238001198s: ('Failed: HTTPSConnectionPool(host=\'eastus-1.azure.cloud2.influxdata.com\', port=443): Read timed out. (read timeout=20.0)', {})
surveyor_worker-1  | [2024-04-15 21:23:01,105: INFO/MainProcess] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[1332d9be-252d-419f-b927-890795f6f42c] received
surveyor_worker-1  | [2024-04-15 21:23:21,281: INFO/ForkPoolWorker-2] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[1332d9be-252d-419f-b927-890795f6f42c] succeeded in 20.17489801600459s: ('Failed: HTTPSConnectionPool(host=\'eastus-1.azure.cloud2.influxdata.com\', port=443): Read timed out. (read timeout=20.0)', {})
surveyor_worker-1  | [2024-04-15 21:30:24,411: INFO/MainProcess] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[7669eb2c-6065-43ea-b864-66e497d88c40] received
surveyor_worker-1  | [2024-04-15 21:30:44,633: INFO/ForkPoolWorker-2] Task performer.task_createBucketDevicesReport.create_bucketDevicesReport[7669eb2c-6065-43ea-b864-66e497d88c40] succeeded in 20.21952486000373s: ('Failed: HTTPSConnectionPool(host=\'eastus-1.azure.cloud2.influxdata.com\', port=443): Read timed out. (read timeout=20.0)', {})
```

krisdthompson, 2024-04-16:
this is fixed. InfluxDB Connection timeout extended from 20 seconds to allow for much larger data sets.

closed 2024-04-16 by krisdthompson