sheeriot/surveyor #46: Spectrum usage by device

Timkipley, 2024-06-03:
Request for spectrum usage by frequency/count chart on the device detail page.  Can be constructed in similar manner to gateway usage by frequency chart on Bucket Report

krisdthompson, 2024-06-03:
Let's start this this bucket device details  page
https://surveyor1.kttex.com/device/bucketdevice/19/gs1/000db5380d79386c/20240527T0500Z/20240603T1812Z/
and this one
https://surveyor1.kttex.com/device/bucketdevice/19/gs1/000db5381078384b/20240527T0500Z/20240603T1812Z/

krisdthompson, 2024-06-08:
sprectrum usage graphs by device are added to the packgraph pages for endnodes and bucketdevices.
spectrum usage is compared to a Channel plan and counted as In Plan or Out of Plan.
feature released.

closed 2024-06-08 by krisdthompson