sheeriot/surveyor #79: Server Error 500 in editing any field on the Device report page

mkhatun1, 2024-07-30:
Receiving the Server Error 500 again, when trying to edit the the "Start" "End" time or "Dev eui" (as we used to do in the past) of the Device level report. 
Here is a device report link (any of the above mentioned action would result the 500 error) : https://surveyor.everynet.com/device/bucketdevice/29/None/mdona1/0c57eb000012cdcd/20240725T0500Z/20240730T1635Z/

krisdthompson, 2024-07-31:
this is caused by the report_group which should NOT be part of the bucket device details form. The Bucket Device has it's group defined by the database (bucket_devices). Removing 'report_group' from the URL and report.

krisdthompson, 2024-07-31:
here is the crash URL on Dev
https://surveyor1.kttex.com/device/bucketdevice/?source=34&meas=mflo1&dev_eui=0c57eb00000a3c24&start=2024-07-30T00%3A00&end=2024-07-31T08%3A02&submit=Submit

krisdthompson, 2024-08-01:
this is fixed and going into QA then Prod

closed 2024-08-01 by krisdthompson