sheeriot/surveyor #92: Missing "tag" field on the Device Uplinks report

mkhatun1, 2024-09-11:
Hi- Could you please add back the 'tag' field on the Device Uplinks report? Mueller is complaining of not getting the device node_id, they need this to identify the device. Here is an example report showing the missing tag column on the Device Uplinks report file: 
https://surveyor.everynet.com/performer/bucketDevicesReport/48/None/moundsview/20240908T0500Z/20240911T1832Z/
[image]

krisdthompson, 2024-09-12:
Here is a test URL for the Dev server.
https://surveyor1.kttex.com/performer/bucketDevicesReport/39/groupA/altavista/20240905T0400Z/20240912T1004Z/
Tag1 and Tag2 are missing.

krisdthompson, 2024-09-12:
fixed. Release pending.

krisdthompson, 2024-09-12:
this is fixed and in production.

closed 2024-09-12 by krisdthompson